rg35xx-cfw/rg35xx-cfw.github.io #136: Drastic occasionally freezes coming out of sleep mode

Biscuits25, 2024-02-15:
When i put the device to sleep mode by pressing power once while drastic is running, it will sometimes freeze when coming out of sleep mode. The drastic menu is unreachable and only pressing the rest button fixes it. I tried the default sleep mode as well as enhanced and instant but it didn't seem to have an effect. I'm also not sure if enhanced and instant are even working as it keeps reverting to default sleep mode. 

Tested with pokemon black and pokemon renegade platinum (romhack).

james-mccloud, 2024-02-16:
Drastic controls works ok
Tanks @acmeplus. If I find some issues I will notify. 👍🏻

Biscuits25, 2024-02-18:
Touch screen controls do seem to work fine but I randomly had an issue with the Yes or No option in Pokemon Platinum. If you select the option to hear all of the game controls, it makes you touch yes or no afterwards. On that screen I can not get the emulated touch controls to select anything. If you select the "no info needed" option instead, it will just make you touch the center of a pokeball, and that works fine. Not sure where the difference is. 

Also really enjoying the os otherwise. Really excited to see where it goes! Appreciate the work,