rg35xx-cfw/rg35xx-cfw.github.io #140: Duckstation not working

igmarsal, 2024-02-17:
When in the PSX emulator, if you select the Duckstation emulator from the application options menu to force scaling or other settings, it fails to start. You need to switch it back to automatic mode for it to work properly

PukeGX, 2024-02-27:
Do you manage to make the standalone Duckstation emulator running ? 
I put all the bios files, but if I want to launch a CHD with Duckstation : no success, even letting any setup to Auto...  tested on RG35XX+ with the 02/13 and 02/20 build

It works Ok with the libretro core.

mhbates, 2024-03-21:
Same here. Anyone able to use Duckstation yet?

PukeGX, 2024-03-22:
Still not working for me.
But with Batocera V40, Libretro PCSX Rearmed has been updated to its last release and is working lot better and is more compatible than the V39 batocera release.

So, in fact, I don't think Duckstation is really usefull now to play PSX in great condition.