rg35xx-cfw/rg35xx-cfw.github.io #143: Wifi connectivity issues for RG35XX Plus

waffledork, 2024-02-18:
The system doesn't connect to wifi upon startup. I have to go to "Network Settings" and turn the wifi off and on again, usually multiple times, in order to get it to connect. The same thing happens when the system comes off of suspend; the wifi doesn't automatically connect.

Additionally, the wifi seems to randomly disconnect while I'm in-game. I'll be hitting retroachievements one minute and the next minute, they won't register on retroachievements.org. If I quit the game, I'll see that the wifi symbol isn't in the upper-right any longer and if I go to "Network Settings", I'll see that the wifi isn't connected anymore.

I went back to the stock OS (latest firmware) this morning to test wifi for it and found that I don't run into any of these issues there. Wifi connects just fine and reconnects when the OS boots and/or is unsuspended and it hasn't dropped in 4 hours so far.

I'd appreciate any help that you can offer!

crudelios, 2024-02-19:
I second this issue, I've been having the exact same symptoms.

adroitful, 2024-02-28:
I have the same issue with the H.. However, I cannot get it to connect to wifi at all. I have played and played with it, but I just can't get it to connect. I will be swapping over to this firmware as soon as this issue is resolved.

acmeplus, 2024-02-28:
Yes, there's a known issue with WPA3 and PMF settings. If your router lets you disable PMF (or 802.11w) then the issue may be resolved. I'm aware stock does not have this issue, but stock is a completely different firmware that implements a different network stack. I don't anticipate WPA3 to be supported anytime soon because 1) I don't have a proper WPA3 routers, and 2) there is a workaround for this issue.

acmeplus, 2024-02-28:
Yeah, you are right, it was there in a previous release but was removed. Added a bullet point to clarify.

adroitful, 2024-02-29:
> Yeah, you are right, it was there in a previous release but was removed. Added a bullet point to clarify.

Thanks! That would have saved me a lot of time.. But, I was determined to figure it out. We would have gotten the answer if it wasn't already a known issue lol

RatchetMiles, 2024-03-11:
I'm not in a position where I can re-config my router :(

Could it be possible to disable WPA3 and force WPA2 on the device itself? Of course, this part is outside my knowledge.

waffledork, 2024-03-11:
@RatchetMiles - my Netgear Orbi mesh router has options to use either WPA2, WPA2+3, and a couple other ones. Check to see if your router allows you to do the same. That said (@acmeplus - you're probably already aware of this), my wifi connectivity has definitely gotten worse after upgrading to v40.

I have a little travel router that I use to provide some security when I'm using hotel wifi - I'm going to set that up with my home network and see if connecting to that helps.

ibriz25, 2024-07-01:
> The system doesn't connect to wifi upon startup. I have to go to "Network Settings" and turn the wifi off and on again, usually multiple times, in order to get it to connect. The same thing happens [...]

ibriz25, 2024-07-01:
i have a issue with my anbernic rg35xx plus  is dpad not working when i play GOD OF WAR  psp game,  and i cant connect wifi , i am using the  latest firmware  is any one here to help me out