rg35xx-cfw/rg35xx-cfw.github.io #149: [ Enhancement/Request ] Include maketools in image

YourWishes, 2024-02-22:
Thanks for your work on this image, just reaching out to see if we can either get make utils so we can compile our own necessary binaries on the device itself, and/or add a compatible pacman mirror that has the armv7l libraries that work on the RG35XX+

Thanks for your effort.

loki666, 2024-02-29:
seccond that, as a more general remak, I'd like this CFW dev to be more open
where is the github with all the changes made to batocera/ES to be able to build this image

It would also allows to have a dedicated issue tracker, and allows other to open Pull Requests

acmeplus, 2024-02-29:
@loki666 To be more open? really? Everything, including toolchains and SDK have been posted here. There are issues, discussions, I participate on discord or reddit when people asks questions. Right where this ticket is, just go one level up to rg35xx-cfw and you will see all repositories, with branches, etc. SDK is in releases too. Frankly there's nothing else that you may need.

acmeplus, 2024-02-29:
@YourWishes the SDK was posted a few weeks ago, you can find it here: https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/releases/tag/rg35xx_plus_h_sdk_20240207

loki666, 2024-02-29:
@acmeplus I didn't want to be rude, or to offend you in any way, sorry.
And certainly not to question your involvement and dedication, you did a great job with this CFW.

Thanks for posting your repos and branches in #124 , I'll certainly try to build an image

My remark was actually to see if I can help, I love my RG35XX H, and If I can fix things here and there, I'd like to open a PR for that.

Again, I apology

acmeplus, 2024-02-29:
No worries @loki666 I've clarified the organization landing page to include the links to the different repositories. 

The batocera and buildroot with rg35xx-wip should more or less build completely without many changes, just clone the batocera.linux repo, switch to the rg35xx-plus-wip, and do the same for the buildroot submodule. Then just do make rg35xx-plus-build and wait for a long time. There may be some hiccups during the build process, so let me know because I've most definitely already seen any possible build failures.

loki666, 2024-02-29:
> No worries @loki666 I've clarified the organization landing page to include the links to the different repositories.
>
> The batocera and buildroot with rg35xx-wip should more or less build completely [...]

I'll give it a spin

loki666, 2024-03-05:
can't build it,
no matter what it's always complaining about a dependency on linux
which makes sense, since since buildroot needs a linux kernel to build