rg35xx-cfw/rg35xx-cfw.github.io #156: Audio cuts out through HDMI

benpage, 2024-02-26:
Using 20240220 on the H.

Plugged into HDMI. Setting audio output to AUTO or AHUBHDMI, audio works in games for about 10-15 secs, then silence. Also, no nav sounds in menu.

Setting audio output to AUDIO CODEC plays through the unit without any issues (including nav sounds).

m8nwalker, 2024-03-01:
Same issue!

mikereg5000, 2024-03-03:
yes me also

benpage, 2024-03-08:
It's possible this is linked to having a bluetooth controller connected (a fair assumption when playing via HDMI)

@m8nwalker @mikereg5000 do you have a BT controller attached when this happens?

I seem to not have the problem if I do the following:
- set audio to HDMI out
- reboot the machine (shut down / start up) while connected to HDMI
- DON'T connect bluetooth controller
- start a game
- HDMI audio doesn't seem to cut out

acmeplus, 2024-03-08:
@benpage interesting finding, is that with the latest release from yesterday?

benpage, 2024-03-08:
@acmeplus no, still on V39 (sorry!)

I did sort of try connecting the BT controller, having the issue, then disconnecting it but I think it still caused the audio issue. Haven't really had time to test out the nuances yet.

m8nwalker, 2024-03-08:
my tests in v40 20240306 version

no bluetooth controller

set audio to hdmi, shutdown, start while connected to hdmi, start a game

the audio sounds through television a few seconds, then the audio cuts, neither tv nor console, sounds returns to tv a few seconds later, the the audio cuts again...

catxmenorca, 2024-03-11:
Hi, i have the same issue.  I have Batocera v40 _batocera-h5-rg35xx-plus-40-20240306.img.gz_

I connect RG35XX H via HDMI with BT controller (ps4) and the sound on tv sounds a few seconds ok, then cuts, and return a few seconds later, and its repeats all time.  I tried with both audio outputs in audio configuration. 

If I use the console without being connected by HDMI and I have BT headphones connected, the sound works correctly all time.

Thanks.

chrizzo-hb, 2024-04-16:
Same issue here.

I am running **batocera-h5-rg35xx-plus-40-20240306.img.gz** on my **RG35XX H**.

To make sure that it is not an issue with my TV or the cable, I tried two different TVs and two different cables, I tried it while the battery was loading via USB and also while it was just running from battery. I also tried it with my bluetooth controller connected and disconnected. It did **not** seem to make the slightest **difference** which combination I was using, the result was always the same:

In every case, I connected my RG35XX H via HDMI and turned it on. After it was booted, background music was playing from the TV or from my device (depending on the previous audio setting), so I went to the system settings and switched audio from _"auto"_ or _"audiocodec"_ to HDMI if required.

Now, whenever I start a game, the audio cuts out - usually after a couple of seconds. Strangely enough, on some emulators (**Gameboy** in particular) the audio **comes back** occasionally, then drops out again. On others, the audio did never come back for as long as I played. To make sure that it **isn't** a **Retroarch** issue, I tried some emulators which are **not** from **libretro**. However **standalone PPSSPP** and **standalone flycast** are **also affected**, even though they seemed to take longer until the sound was cutting out.

Strangely enough, **ScummVM** seems **not** to be **affected**! I was playing some "Day Of The Tentacle" and some "Broken Sword" for a couple of minutes without any issues. (I played them with the built-in controls or my regular gamepad, no other device was connected.)

On a couple of occasions, audio was **cut out** while I was browsing the **emulation station** game lists, so the issue might not be limited to the emulators.

Whenever the issue occurs, I can "fix" it for a while by going back to the system settings. There I can switch from HDMI back to codec, which makes the audio play from the device again. If I then switch back to HDMI, the audio will run through HDMI again. But as soon as I start another game, audio will be cut off after a few seconds.

I do **not** have these issues when I play directly on the handheld without HDMI.

uzumak11, 2024-04-17:
The next version seems to have fixed hdmi. Wait for Knulli to hit public release