rg35xx-cfw/rg35xx-cfw.github.io #157: Connecting to NAS

seanstardev, 2024-02-27:
Trying to connect my NAS to the Plus. Added the below to the batocera.conf file, but don't think it is working:

sharedevice=NETWORK
sharenetwork_smb1=LIBRARY@192.168.0.2:DiskStation:username=Me,password=pass123,vers=1.0

... I say not working because when I go to the library folder in userdata, using OD-Commander,  I am taken to the standard sd card library folder.

I can ssh in and verify that the SMB is set up and working, and browse any folders etc that I like. For anyone curious, this is the instruction:

smbclient \\192.168.0.2\DiskStation -U Me --option='client min protocol=NT1'

waffledork, 2024-02-28:
Out of curiosity, why are you trying to do this? Is the goal to run games that are stored on your NAS and not on your SD card?

acmeplus, 2024-02-28:
@seanstardev it's probably related to some issues regarding mounting devices. Hopefully once those are solved that will work automatically.

seanstardev, 2024-02-29:
> Out of curiosity, why are you trying to do this? Is the goal to run games that are stored on your NAS and not on your SD card?

I want to set my devices up so that they pull roms from the NAS (a much bigger repo) when available. Otherwise - just pull from the smaller collections on sd cards. 

Now... I don't think Batocera allows for multiple rom directories for a system yet, but this is a cool step in that direction :)