rg35xx-cfw/rg35xx-cfw.github.io #158: Using both SD Cards

seanstardev, 2024-02-27:
I believe this is a known issue, but while it is possible to mount a partition on the 1st SD card, it is not possible to save the overlay. As such, the drive needs to be mapped each time the system is restarted.

I have personally verified this machine can read at least 1TB SD cards, so the prospect of this machine having 2TB of storage space with Batocera is very exciting :)

acmeplus, 2024-02-28:
There are still issues regarding mount points that are going to be fixed in a new release. Those issues are part of what's preventing batocera mount points to work, and also would probably allow for users to mount two different locations.

seanstardev, 2024-03-10:
> There are still issues regarding mount points that are going to be fixed in a new release. Those issues are part of what's preventing batocera mount points to work, and also would probably allow for users to mount two different locations.

A quick update here, having tried the v40 version.

If I ssh in and run the following:
`mkdir -p /mnt/Roms2 & mount -t exfat /dev/mmcblk0p4 /mnt/Roms2`

... I can successfully mount and map the storage partition of the first SD card. (Obviously - I know that my storage disk is called '/dev/mmcblk0p4' here, and set my mapped folder name to be 'Roms2'). This works, but is forgotten again on reboot.

So I then tried adding a 'boot-custom.sh' script on the Batocera partition's boot folder. This didn't work, so I also tried renaming the script to 'postshare.sh' and rebooting. This didn't work either.

My script is below - I'm not sure it is written correctly, I must admit. But any further guidance would be appreciated. Also - I hope this helps someone get farther than I :)

The script:

```
UUID="DE1B-1DB7"

# The mount point for the device
MOUNT_POINT="/mnt/Roms2"

# Find the device by UUID
DEVICE=$(blkid -U $UUID)

# Check if the device was found
if [ -z "$DEVICE" ]; then
    echo "Device with UUID $UUID not found."
    exit 1
fi

# Check if the mount point already exists, if not create it
if [ ! -d "$MOUNT_POINT" ]; then
    echo "Creating mount point: $MOUNT_POINT"
    mkdir -p $MOUNT_POINT
fi

# Mount the device
echo "Mounting $DEVICE to $MOUNT_POINT"
mount -t exfat $DEVICE $MOUNT_POINT

echo "Mount successful."
```

WorstGit, 2024-03-29:
You need to have both data partitions on both cards formatted to ext4 (the Batocera will format its self and just leave it alone.
This is possible with Windows with some software (Paragon worked for me) else use a Linux distro to format to ext4.

Insert the firmware card into the device and boot.
Shut down and insert the second card and boot.
Now you've got both cards populated with Batocera's folder structure whilst formatted to ext4

SSH into the device, run lsblk to check the drive IDs and grab the ID of the large partition on the boot device. 
From there cd /boot to enter the boot directory and create a file named boot-custom.sh
This file should contain

mkdir /userdata2
mount /dev/ <your drive ID> /userdata2

Boot your device and in OD Commander (ports) you should see both userdata and userdata 2
If not, reboot again and/or check steps.
Choose which rom folders you want on which card. For example all PSX on the same card as Batocera, delete the PSX folder from userdata (this is the ext card) and create a symlink of PSX from userdata2 to userdata.

Confirmed working on latest release, Batocera V40 Beta.

All credit to and further detail to
https://www.reddit.com/r/RG35XX_Plus/comments/1av4evt/dual_sd_card_setup_batocera_koriki/

seanstardev, 2024-03-29:
> You need to have both data partitions on both cards formatted to ext4 (the Batocera will format its self and just leave it alone. This is possible with Windows with some software (Paragon worked for me) else use a Linux distro to format to ext4.
> 
> Insert the firmware card into the device and boot. Shut down and insert the second card and boot. Now you've got both cards populated with Batocera's folder structure whilst formatted to ext4
> 
> SSH into the device, run lsblk to check the drive IDs and grab the ID of the large partition on the boot device. From there cd /boot to enter the boot directory and create a file named boot-custom.sh This file should contain
> 
> mkdir /userdata2 mount /dev/ /userdata2
> 
> Boot your device and in OD Commander (ports) you should see both userdata and userdata 2 If not, reboot again and/or check steps. Choose which rom folders you want on which card. For example all PSX on the same card as Batocera, delete the PSX folder from userdata (this is the ext card) and create a symlink of PSX from userdata2 to userdata.
> 
> Confirmed working on latest release, Batocera V40 Beta.
> 
> All credit to and further detail to https://www.reddit.com/r/RG35XX_Plus/comments/1av4evt/dual_sd_card_setup_batocera_koriki/

Well, thank you very much indeed for taking the time to reply.

It's all so confusing. It seems ext4 formatting is only needed for the boot-custom.sh to be able to work. Even then - I don't see why the sh file is needed since batocera-save-overlay is meant to be working now.

(ext4 is 100% not needed if you just map the paths through ssh, but of course, that mapping is lost on reboot). 

Well done for getting this working, but yeah... ext4 is a bridge too far for me. I hope next bato release offers a more straight forward solution.

Thanks again.

WorstGit, 2024-03-29:
ext4 is required for symlink as far as I know. I tried this method previously while using exFAT on my partitions and it didn't work. I'm using dual 64gb cards, which made this quite necessary.

If you do decide to try ext4, a third party program does work but for longevity I'd recommend using a USB Linux distro. I tried fiddling through WSL but it was more headache than it was worth.
Also, I found ancient ext2/3/4 windows drivers online, it's a rabbit hole not worth going down.