rg35xx-cfw/rg35xx-cfw.github.io #161: Missing NDS menu and weird game controller issue

Andrewbestman, 2024-02-27:
It’s my first time writing here and thanks a lot for the custom firmware first of all. ❤️

After updating Batocera to the 20/2 build all NDS games are not accessible while all bios and game files are here. Tried updating the game list multiple times and also show all game consoles with no game but NDS is just gone.
Also the controller that I’ve connected over Bluetooth only works in the menus but not in the games even after manual mapping the controls.

igeljaeger, 2024-02-29:
same problem here. It's like nds path scanning gets skipped

acmeplus, 2024-02-29:
See this post for a temporary workaround:
https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/discussions/154#discussioncomment-8633599