rg35xx-cfw/rg35xx-cfw.github.io #162: ECWolf disappeared

PukeGX, 2024-02-27:
Hello ! First, thanks for your great work porting Batocera on these device.
I use a RG35XX+

With the last 20240220 the ECWolf menu disappeared and so, the games aren't playable.
The ecwolf directory wasn't present in the roms folder created by this version too, so i re-add it, but the menu doesn't show

This menu was showing and the games worked great in the release from 2024/02/13.

acmeplus, 2024-03-03:
Most likely it was left out from the compilation configuration.  I’ll re-add it again for the next release

acmeplus, 2024-03-07:
Just added libretro-ecwolf to the build, it should be in the next release.

PukeGX, 2024-03-08:
Thanks ! 

Because I just test the v40 and it still not here.
Just waiting for next release 

Thanks a lot for your work !