rg35xx-cfw/rg35xx-cfw.github.io #167: Retroachievement not connecting

MaxAndre1102, 2024-03-02:
I tried to connect to the Retroachievement and I could get the trophies I have, however when I try to connect to a new game, there is a error message saying saying that there is a issue with the username or password. Is that happening to someone else?

acmeplus, 2024-03-07:
Please give it a try with the latest firmware, there are a lot of changes and that may have been fixed in upstream.

MaxAndre1102, 2024-03-07:
> Please give it a try with the latest firmware, there are a lot of changes and that may have been fixed in upstream.

I tried the 20240207 version and didnt work. I could download my achievements from previous systems but when I try to connect to a new game says that there is an issue to connect.

MaxAndre1102, 2024-03-09:
Issue fixed. I created a new account and was able to connect. something wrong with my old account.

closed 2024-03-09 by MaxAndre1102