rg35xx-cfw/rg35xx-cfw.github.io #168: RG35XX H - Always detecting GAMEPAD#2 upon turning it on, and pressing power

totohenzo, 2024-03-04:
When I turn the unit on, i'm greeted with a prompt to set-up "Gamepad#2". and whenever I press the power button it does it again. I'm using the latest Feb 13 2024 update. i tried formatting and installing the Feb 8 2024 version instead, but i still get the same prompt. I recently bought two units, one for me and one for my daughter. both are having the same issue

acmeplus, 2024-03-07:
Will check, I've done most of the testing of this release on the Plus and not on the H, but there are some input key differences that may explain that.

totohenzo, 2024-03-08:
I removed the 2nd SD card from the second slot (which had stock OS btw), and it's not having the issue anymore! :) Sadly, i can still get it even after removing or renaming the system folder. But it's fine. I'll use my RG35XX H with just one sd card with batocera OS in slot one. THANKS FOR THE REPLY!!!