rg35xx-cfw/rg35xx-cfw.github.io #170: "Power surge on the usb port" Device won't turn on

ArmoMike, 2024-03-06:
I know this is not a batocera issue but i was hoping somebody had the same issue and could help me.. I've been charging my rg35xx plus with my pc since i got it and everything has been fine . Yesterday i tried charging it with a powerbank and now all of a sudden the device won't turn on or charge anymore . I plug it into my pc and i get the error : "Power surge on the USB port Unknown USB Device needs more power than the port can supply". I tried unplugging the battery and waiting , holding the power + reset button , charging with other plugs/chargers but nothing. Is this thing dead ?

Tartopoireaux, 2024-03-06:
I had same issue, battery is dead. You can try to put the original sd card and plug the usb port. You will see battery charging and the device swtich off...
I think this device needs low power for charging.

acmeplus, 2024-03-07:
I've had issues with the pc usb port not feeding enough power to charge, and yes, once it gets below a certain threshold the only way to boot is to connect it to a regular power brick. So if you have that issue that should do the trick.

ArmoMike, 2024-03-07:
> I've had issues with the pc usb port not feeding enough power to charge, and yes, once it gets below a certain threshold the only way to boot is to connect it to a regular power brick. So if you have that issue that should do the trick.

nah it wont turn on with anything sadly :(