rg35xx-cfw/rg35xx-cfw.github.io #171: Official Pico-8 not working.

ianmacd, 2024-03-06:
I've tried adding v0.2.6b of the official Lexaloffle Pico-8 distro for the Raspberry Pi to your 20240213 and 20240220 Batocera builds according to the [Batocera wiki page on the subject](https://wiki.batocera.org/systems:pico8), but sadly to no avail.

I'm running a two-card set-up, with nothing but your OS image in the first slot, and all of my BIOS and ROM files in the second.

After configuring, I can select the Lexaloffle emulator in the UI, but games don't start. I just get dropped back to the Emulation Station Pico-8 game selection screen. Neither the 32- or 64-bit ARM binary works. Perhaps they're failing on some dynamic library that doesn't ship with Batocera; that would be my guess, anyway, but I'm not sure where to look for logs.

It would be great to be able to use the official Pico-8 binary, as opposed to Retro8.

acmeplus, 2024-03-07:
I'll check. Frankly I've not tested it after the full upstream merge so I'm not surprised if some things fail.

ianmacd, 2024-03-07:
> Instead you need to copy `pico8_dyn` to `pico8`.

Yes, that was the key. It works now.

I even knew I was running the Pi version before, but I thought that was also the correct binary for these devices, given their CPU architecture.

Many thanks for the prompt solution.

acmeplus, 2024-03-07:
> I even knew I was running the Pi version before, but I thought that was also the correct binary for these devices, given their CPU architecture.

I had the same exact thought, I knew it was the rpi version but was assuming batocera was using pico8_dyn directly... Anyway, glad that it works.

dbdrwdn, 2024-03-24:
> I've just checked, and there's an error in the batocera wiki page about that. The issue is that it tries to run pico8 and that's the raspberry pi version. Instead you need to copy `pico8_dyn` to `pico8`.
> 
> Basically if you follow the instructions:
> 
> * Copy the extracted pico8 to bios/pico-8
> * In that folder copy pico8_dyn to pico8
> * Copy the es_systems_pico8.cfg to system/configs/emulationstation/
> 
> Let me know how that goes.

I did just that and ended up with loading into a black screen and after around a second back to the menu.
Is there any log I can check?

Leaver90, 2024-04-06:
> I did just that and ended up with loading into a black screen and after around a second back to the menu. Is there any log I can check?

That is exactly what is happening to me and I'm still stuck.

Leaver90, 2024-04-06:
Nevermind I figured it out. I'll include more detail below:

Step 1: Extract the pico8 folder to to bios folder within SHARE
Step 2: Rename the original pico8 file to pico8.bak
Step 3: Copy and paste pico8_dyn within the same folder and then rename the copied file to pico8
Step 4: Navigate out of your current folder (bios) into the main directory
Step 5: Follow this path, system/configs/emulationstation
Step 6: Create a file called es_systems_pico8.cfg (easiest way is to right click and make a txt file and just rename it to that file name)
Step 7: Open that file using a basic text editor, like notepad
Step 8: Paste the following in

`<?xml version="1.0" encoding="UTF-8"?>
<systemList>
  <system>
        <name>pico8</name>
	<emulators>
            <emulator name="lexaloffle">
                <cores>
                    <core default="true">pico8_official</core>
                </cores>
            </emulator>
            <emulator name="libretro">
                <cores>
                    <core>retro8</core>
                </cores>
            </emulator>
        </emulators>
  </system>
</systemList>`

Step 9: Make sure you have splore.p8 in the right spot. I have it in the bios/pico8 and in roms/pico8. I'm sure only one was needed but hey, it works.

Step 10: Eject your SD card and pop it in your device
Step 11: Navigate to the PICO-8 gamelist and press select, go to Advanced System Options, and select LEXALOFFLE: PICO8 OFFICIAL as the Emulator. 
Step 12: Let er rip tater chip

froznice, 2024-05-13:
> Nevermind I figured it out. I'll include more detail below:
>
> Step 1: Extract the pico8 folder to to bios folder within SHARE Step 2: Rename the original pico8 file to pico8.bak Step 3: Copy and [...]

I have done all of these steps, still can not get games to open. I am on v40 and have converted my share partition to ext4. Would this change anything?