rg35xx-cfw/rg35xx-cfw.github.io #172: thanks again

loki666, 2024-03-07:
thanks again for not updataing completly your batocera fork
this time, it's missing the correct boot.img (with squashfs module)
don't worry I took it from your image

honestly, it looks like your are intendedly hiding stuff so others can't build the image by themself, so in a way you are not really better than Anbenic reguarding open source licenses

acmeplus, 2024-03-07:
Frankly your comment is absolutely uncalled for, rude, and throws away all the effort I’m putting trying to help the community. 

Yes, I forgot to add the boot.img and no, it was not intentional. I do this in my free time and I try to share everything and keep everything up to date. 

With comments like this it really questions whether it’s worth doing this at all. “Thanks again” as you say.

jdub850000, 2024-03-07:
@acmeplus Agreed. The comment was utterly rude and uncalled for. 
I and I'm sure 99.999% of this community are very supportive and appreciative of the great work you are doing.

loki666, 2024-03-07:
that was rude indeed

but you don't seems to be open to help from people (like me) that actually wanted to help this cfw
like there is no way to open issues on your batocera fork,
you ignore pull requests
and you didn't responded to my other comment here https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/issues/149#issuecomment-1978198438

so yes, you are doing a great work here, but you are missing the extra mile to be really open, and yes, it makes me question why

I'm also doing this on my free time, and I've actually started my own fork (that is compiling ;) using what I could salvage from yours

I'm not there yet, but at lease issues it compiles and boot

acmeplus, 2024-03-07:
> that was rude indeed
> 
> but you don't seems to be open to help from people (like me) that actually wanted to help this cfw like there is no way to open issues on your batocera fork, you ignore pull requests and you didn't responded to my other comment here [#149 (comment)](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/issues/149#issuecomment-1978198438)
> 
> so yes, you are doing a great work here, but you are missing the extra mile to be really open, and yes, it makes me question why
> 
> I'm also doing this on my free time, and I've actually started my own fork (that is compiling ;) using what I could salvage from yours
> 
> I'm not there yet, but at lease it compiles and boot

There are zero pull request open on this firmware, no one has opened one yet, so I don't know where you come up with that idea that I ignore pull request. 

[image]

You say there's no way to open issues on my batocera fork. The issues are here in this repo and not on the individual repos (batocera.linux, buildroot). This project is organized as a GitHub organization and not as individual repos. It's not perfect but it helps me organizing the work and is the way I've done it for other firmwares. 

Tour message from the other day (https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/issues/149#issuecomment-1978198438) only said that you had issues compiling but you didn't post the error message. I have limited time and I cannot guess every single ticket what is about. As a matter of fact, someone on discord posted the same error you mentioned, and I fixed it in the repo: 

[image]

I'm happy you have your own fork, if you don't like the way I manage the project, do it your way. But please don't make unfounded accusations about things before asking questions.

alpgarcia, 2024-03-07:
This: https://github.com/rg35xx-cfw/batocera.linux/pull/1/files

neocvera, 2024-03-07:
You’ve been doing this long enough to know there’s always going to be dipshits bitching and moaning and/or pretending to know things. You’ve held a stance of ensuring everything is out there in the open for others to participate. I appreciate everything you have contributed.

Kilvoctu, 2024-03-09:
> With comments like this it really questions whether it’s worth doing this at all. “Thanks again” as you say.

Comments like this make me feel the same way.
I used to be very excited and passionate about one of my project until people came along to share their opinions and demands in the most rude way possible, as if it's my job and they're my clients. Wasn't fun for me so I straight up stepped away from the project (as its sole developer). Modestly sized repo, close to 300 stars.  
~1 year later, kinda back on it, at 2% of the motivation and a "it'll be done when I feel like it..." schedule.

The stuff you're doing here is outside my field of knowledge, so I appreciate what you do @acmeplus . A genuine thank you from myself and many others. What you do makes a positive impact on so many people; hope it reminds you that it's worth it.

Surprised this "issue" hasn't been closed. It adds nothing but bad vibes

loki666, 2024-03-09:
issue was closed because kernel image is still missing

alpgarcia, 2024-03-09:
All in all, the title and description are not informative, and the language is aggresive.

A proper issue should be informative, with requests instead of demands, not ironic, and without references to other people life's or intentions. You are not required to help, but please respect others.

So please, close this issue and open a proper one.

Thanks.

closed 2024-03-09 by loki666