rg35xx-cfw/rg35xx-cfw.github.io #175: Jaguar items not displayed

Wolvbob, 2024-03-07:
Despite there are ROMs in the "jaguar" folder, which was correctly displayed in previous version, with the Batocera V40 dated from the 6th of March, system is not displayed.

I tried to build a file like for the pico8 (with the format es_systems_<custom_name>.cfg) but it does not work. I tried also to builf a file named "es_systems.cfg" with jaguar system described inside but even worse : the system starts with an error message.

Thanks a lot and great job done on this V40 version ! Well done !

acmeplus, 2024-03-07:
Looks like virtual jaguar was not compiled for this version, I'll add it for the next one.

Wolvbob, 2024-03-08:
Thanks a lot for your quick answer and the great job you do on your free time for this device.