rg35xx-cfw/rg35xx-cfw.github.io #176: Snes9x is not "Current" and Supafaust crashes

WaldyTMS, 2024-03-08:
The Snes9x core this uses is one of the "year" versions of either 2002, 2005, or 2010. The reason I know this is because the music quality is nowhere near as accurate as "Current," and I've used the "year" versions on other handhelds before enough to know the difference. Is there any way to update the Snes9x core to be the "Current" version? Also, when selecting Supafaust as the emulator for SNES, the game always crashes.

seanstardev, 2024-03-10:
> The Snes9x core this uses is one of the "year" versions of either 2002, 2005, or 2010. The reason I know this is because the music quality is nowhere near as accurate as "Current," and I've used the [...]

I couldn't verify this. I tried by going into:

'Game Settings > Per System Advanced Configuration > Super Nintendo Entertainment System > Emulator'

The only options are:
- SNES9X
- SNES9X NEXT

... I don't see options for 2002, 2005, 2010 versions, etc.

Not saying you're wrong, by any means, Specific core versions aren't shown, which is annoying. But nothing explicitly states we're using a year version. I also don't see Supafaust as an option at all.

rendar82, 2024-03-13:
Core version info can be turned on via the Retroarch User Interface settings, and yes the SNES9X core in this release is version 1.62.3-ec4ebfc (which is the most recent version of SNES9X core, released March 2023).

As can be verified via the Libreto core release page: https://github.com/snes9xgit/snes9x/releases or found with a simple Google search regarding SNES9X cores, the SNES9X Core Version 1.62.3-ec4ebfc being used in Koriki Batocera IS the "SNES9X current" proper Libreto core. 

This is not one of the "SNES9X 2002", "SNES9XX 2005", or "SNES9X 2010" cores, which are just snapshots of what the SNES9X emulator code looked like in those years. As the core developed, emulator accuracy increased but so did the system requirements required to run games full-speed. A great example of the classic accuracy & new features vs. raw performance trade-off that is ever present in emulation.

Music/Sound quality in emulation can be affected by a literal ton of things aside from the actual quality of sound reproduction output by the emulation core, including the fact that both the RG35XX-Plus (mono audio) and RG35XX-Plus-H devices are made to be as cheap and affordable as possible and the audio chipsets and speakers/audio quality are relatively poor to begin with (but decent considering the cheap price of these tiny devices).