rg35xx-cfw/rg35xx-cfw.github.io #178: Sound output Issue when plugging HDMI

Shakyhandsmcg, 2024-03-08:
RG35xx plus - I had been using the 13.02.24 build and the sound was fine but when i updated to the 20.02.24 build now the 'Auto' Setting on Audio output doesnt work - It will continue to output to the device until you go into system settings and change it to HDMIhub for HDMI output or Codec (something?) for device output.

Shakyhandsmcg, 2024-03-13:
Thank you! yeah this works and the frontend sound will be fine through hdmi but the game sound then experiences a weird issue where it will be fine for a couple minutes then just cut in and out but its off way more than its on.

I found this happens with all systems as i initially thought maybe was just PS1 or that i had a bad rom.

acmeplus, 2024-03-13:
Yeah, it's still a problem. Some users reported that changing the audio device or profile fixed it but it fails again. I'm still investigating this.

loki666, 2024-03-20:
@acmeplus I'll give you a hint

```fbset -g 1280 720 1280 1440 32 -t 13426 192 56 22 1 136 3```

should also fix the terrible input lag

madlogik, 2024-03-26:
> @acmeplus I'll give you a hint
> 
> `fbset -g 1280 720 1280 1440 32 -t 13426 192 56 22 1 136 3`
> 
> should also fix the terrible input lag

is this something I can test out if I have the current v40 beta or does it have to be hardcoded? 
I do suffer from that intermitant hdmi audio output. :P

loki666, 2024-03-27:
you can run it from a shell when connected to HDMI, it will switch the frequency to 60Hz instead of 22Hz which probably alot of the TV set have issues to sync with

madlogik, 2024-03-27:
> you can run it from a shell when connected to HDMI, it will switch the frequency to 60Hz instead of 22Hz which probably alot of the TV set have issues to sync with
> 
> see https://wiki.batocera.org/access_the_batocera_via_ssh

just had a chance to test. no. does not prevent the issue nor fixes it immediately. 
Thanks for sharing. The video never has an issue by the way. And my tv sees the input as 720p @ 60fps all the time. 

Homefuly we get a fix in v41 and hopefully OTA :D

loki666, 2024-03-28:
well `fbset --info` clearly shows that the pixeclock is still the one used for 640x480@60Hz, and since users reported that sound over HDMI is working only a third of the time, the maths shows that you need +/- 3 times the pixel rate for 1280x720@60Hz

now what reports `fbset` could be completely unrelated to the actual pixelrate set by the kernel driver.

I've tried to follow what's going on in the driver, but I got lost after the 4th level of function pointers forwarding used when switching from LCD to HDMI, there are codes thats changes the clock based on the video mode, but I have no clues if they are triggered or not

real or not pixelrate, without the timing info for `fbset`, there is clearly an increased lag input

madlogik, 2024-03-29:
FYI: on V40 beta: I can get stable HDMI audio output if I use a retrolib core and go in the audio settings from the retroarch menu 
(fn+b / back / settings / audio / output / device / change from default to (sysdefault:CARD=ahubhdmi) ) 
This way the audio stays on. I can save that setting core by core... but it only works in that retroarch core, and would not work at all on non retrolib cores) . 

If I try to change the output system wide by going in batocera's main menu then : (start / system settings / audio output) and change it from AUTO to AHUBHDMI then the sound would come in for a few seconds, then out for about a minute or so and randomly so.. when it happens, I actually have no audio devices at all, assuming the entire audio sink crashes and then gets restarted after hanging for a while. 

Also noticed that none of the n64 cores (rice, mupen and gen2 and the liretro ones work over hdmi) I default to rice otherwise while in handheld mode ... but they wont start at all over hdmi :D I'll post that in another thread if it'S not already posted. 

thanks again @acmeplus  for all your hard work, this device (rg35xx-h) got a lot of my friends interested and I just got 3 consoles from ali and awaiting my sd cards tomorrow for them, ... I want to help them out getting this amazing handheld setup and your amazing port of batocera for it and all the screenscrapers work for all those screenshots / videos / manuals and other pieces to make it a truly GREAT experience. 

That being said, i'm the geek and they are not and ideally I would like not to have to support their problems with it if any in the future...so I may as well work now to iron  out the kinks with you guy as I am a fan of that setup and daily drive it myself. 

oh I also need to check to see if anyone reported the screensaver not working properly to skip videos with a gamepad's dpad. :D (at least while connected to hdmi)...

loki666, 2024-03-29:
for what it worth, I don't seems to have these kind of issues with my fork of v39 (based on acme changes)
https://github.com/loki666/batocera.linux/releases/tag/rg35xx-plus-v5