rg35xx-cfw/rg35xx-cfw.github.io #180: Analog stick mapping issue

Jumby46, 2024-03-09:
I tried to remap the analog stick to the D-pad though quick menu (for PSP and Dreamcast), but the mapping gives no results. Same for N64 but in this case I was unable to access to the configuration page.

Am I doing something wrong or it is a known issue?

Thank you

astapenkoav, 2024-03-11:
@Jumby46
If you about RG35XX then

1.Go to Controllers and Bluetooth Settings -> Controller Mapping.
2. Hold button A to define the controller .
3. Set the keys North (X), south(B), west(Y), east(A). Up arrow, down arrow, left arrow, right arrow.
4. But for analogue keys (those that do not exist) just hold any already assigned button for a long time and the setting of this key is skipped. Make it for all analog buttons.
5. Save.
6. In P1 controller select #0 DEEPLAY-Keys just in case.

And then it worked in N64 games well, tested in Paper Mario and Yoshi's Story.

Jumby46, 2024-03-11:
Thank you so much for the reply.

Yes I’m talking about RG35XX Plus. 
This solve the issue with the N64 but it is still impossibile to remap the D-Pad for Psp and Dreamcast.
Any hints to fix these consoles?

closed 2024-03-11 by Jumby46

astapenkoav, 2024-03-11:
@Jumby46