rg35xx-cfw/rg35xx-cfw.github.io #185: Can't find Nintendo DS

joseparreiras, 2024-03-12:
I am very newbie to all this. That is the first RHH that I own so I am probably having basic issues with the console. I got the RG35XX plus yesterday and, after watching a couple of youtube reviews I decided to install the Batocera CFW. I backed up the original SD that came with the device, installed the system according to the guide and booted it up. After the SHARED partition was created, I copied both the bios and roms that came in the console. That included a bunch of other consoles that I know are not supported by Batocera in this device, but it also contained some ROMS and BIOS for Nintendo DS using Drastic that I wanted to play. When I booted the device again, I could not find the option for NDS on the main menu, though the bios and roms are indeed in the SD card. I tried to update the game library but it still did not appear on Emulation Station.
I read the entire guide and looked up on the discussion page for some walkthrough on how to get DS working on the device, but I am still lacking a clear guide on that. From the reviews that I saw previously, NDS was actually plug and play after installation, so I don't know what should I do to get it working. 
Some silly issues that I thought could be causing this are:

- What path should I put my roms for the device to be able to find it (also can't seem to find the ones for SNES)
- Do I need other drastic bios rather than the ones coming with the stock SD card?

Also some other things that have been bothering me:

1. I am having to set my hotkeys on RetroArch everytime I open a game, I saw there is an option for saving the config, but it doesn't seem to be working.
2. As mentioned, SNES appears on Emulation Station, but it can't scan the games I put on my SD card. Which is the correct path for SNES?

Thank you for your work on this CFW. I am having a great time playing the other systems that I have long awaited for.

acmeplus, 2024-03-12:
Make sure you install v40: https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/releases/tag/rg35xx_plus_h_batocera_20240306

The NDS was broken in V39. In V40 it should work without any changes required and the roms/nds folder will be there.

If you had a previous install, make sure you remove or rename the system folder so it gets recreated the next boot. 

About hotkeys, batocera controls the management of the keys because it does that so the controller mapping in batocera works. You can redefine the hotkeys manually, but that has to be done via configuration. See this guide for details: https://wiki.batocera.org/advanced_retroarch_settings?s[]=hotkey

SNES games are in the roms/snes folders, it may be that the extension of your roms is not compatible. These are the compatible extensions: [smc, fig, sfc, gd3, gd7, dx2, bsx, swc, zip, 7z]

joseparreiras, 2024-03-12:
Thanks @acmeplus
I think I was just unlucky that a new release was made so quickly after I downloaded the previous one that I did not noticed. I updated to v40 and NDS is working greatly now.
The main hotkey change I wanted to apply is to switch `Menu + D-Pad Right` to Fast Forward (Toggle) instead of (Hold). I will use the doc you sent to see if I can change it.
Thanks again!

closed 2024-03-12 by joseparreiras