rg35xx-cfw/rg35xx-cfw.github.io #186: Pico 8 ghost inputs

danieldflip, 2024-03-13:
Hi, every time I play on Pico-8, it seems like there are some hiccups here and there where the right input goes on hold, making the character go right and keeps on going right until I pause the game. I found this issue on v39 and v40 as well.

acmeplus, 2024-03-13:
Will check. Is this with the Plus or H?

danieldflip, 2024-03-13:
> Will check. Is this with the Plus or H?

It’s on both. Recently got the H and I thought it had stick drifting right off the bat, but I tried it on my plus and it did the same thing. I was playing Celeste and Buns: Bunny Survivor for reference. Both games were doing it.

nlflint, 2024-05-21:
Ditto on the d-pad getting stuck in certain directions. I can repro it consistently.

Repro 1: (right movement)
1. Start Celeste
2. Start first level
3. Quickly slide thumb right to left over dpad. As fast as you can, may take a few tries.
Expected: Character hasn't moved much.
Actual: Character walks right continuously while no inputs are pressed.

Repro 2: (down movement)
1. Start Celeste
2. Start first level
3. Quickly slide thumb down to up over dpad. As fast as you can, may take a few tries.
Expected: Character hasn't moved much.
Actual: Character is continuously crouching, while no inputs are pressed.

Specs:
Hardware: RGB35xx Plus
OS: Batocera v40
Emulator: Pico-8 (bug reproduces on official emulator and Retro8
Game: Any game.