rg35xx-cfw/rg35xx-cfw.github.io #189: Retroachievements ?

Tryum, 2024-03-16:
I read from other issues it should be working (as well as from the v40 changelog)...  
However, if setup is working Ok, when I launch a game it doesn't trigger retroachievements.  
If I access the retroarch menu, close the game and go to settings, I see retroachievements being disable.  
If I toggle them on, the configuration I set in ES is there. Unfortunatly, that doesn't save for the next launch.

uzumak11, 2024-03-18:
Are you enabling them in retroarch alone? I don't think that is meant to work. 

In batocera UI, under game settings > retroachievements. Did you do the setup here?

Tryum, 2024-03-18:
It is indeed configured in ES (Batocera), and I can check my account there.
But in retroarch, while configured ok (with the settings from ES), the feature is disabled.... and not save upon exiting and relaunching a game.

uzumak11, 2024-03-18:
It is also dependant on which emulator/core supports achievements. Make sure the emulators chosen per system are as listed in the link below: 

https://docs.libretro.com/guides/retroachievements/

Tryum, 2024-03-18:
I'm trying with NES, default emulator being FCEUmm which is list as supported.
I also tried with snes9x with the DK homebrew game, same result.

One new thing I noticed : I fond a way to enable retroarch retroachievements without closing the game, but if I try to check the achievements from retroarch quickmenu, it crashes back to Batocera.

uzumak11, 2024-03-19:
> I'm trying with NES, default emulator being FCEUmm which is list as supported.
>
> I also tried with snes9x with the DK homebrew game, same result.
>
>
>
> One new thing I noticed : I fond a way to enable [...]

Cannot reproduce this. Achievements pop up normally. Did you index the games you have after enabling it in ES. 

Last resort, maybe deleting system folder and letting the OS recreate it is an option.

Tryum, 2024-04-07:
Hi,
Some update about this issue : I got the same trouble with batocera on x86... figured out that the issue happens here : https://github.com/batocera-linux/batocera.linux/blob/46567dbea28e84977bca22e2db01bb954e6478a7/package/batocera/core/batocera-configgen/configgen/configgen/generators/libretro/libretroConfig.py#L62
I have a firewall block to 8.8.8.8, resulting in a false "no internet connection" detection, resulting in achievements forced to false 😭

Tryum, 2024-04-12:
Will be fixed in batocera v40

closed 2024-04-12 by Tryum