rg35xx-cfw/rg35xx-cfw.github.io #190: Really Slow Bootup?

rebeltaz, 2024-03-18:
I have tried a few different OS's for the rg35xx h and batocera is far and away the best, so please don't get me wrong. I will keep using this over the others, but... why does batocera take SO long to boot? All of the other OS's, including the OEM, take just a few seconds to fully boot. Botocera, though, takes a full one minute and 15 seconds to go from the original splash screen (the one with all of the console logos) to the 'batocera' screen (where it actually looks like it's scanning the games and settings). 

I am running the latest (20240207) version of batocera. I  currently have 1012 games installed. Each OS is tested using the exact same micro sd card - not a similar card or another card, but the SAME card. 

Is there anything that I can do to speed up the boot time on this?

uzumak11, 2024-03-18:
That isn't the latest build. You need the v40 version, released on march 6. With that, and formatting my cards as ext 4 (don't know if this helps as such), boot time is around 20 seconds.

rebeltaz, 2024-03-18:
> That isn't the latest build. You need the v40 version, released on march 6. With that, and formatting my cards as ext 4 (don't know if this helps as such), boot time is around 20 seconds.

I don't know why I didn't see that newer release when I looked, but I do see it now. I will definitely try that. Thank you!

meals12345, 2024-03-19:
Would that be just a regular batocera download for v40

rebeltaz, 2024-03-20:
> That isn't the latest build. You need the v40 version, released on march 6. With that, and formatting my cards as ext 4 (don't know if this helps as such), boot time is around 20 seconds.

I wanted to thank you. That REALLY sped things up.

closed 2024-03-20 by rebeltaz