rg35xx-cfw/rg35xx-cfw.github.io #191: Gamepad 8BitDo M30 and ather does not connect to Batocera

AlexLeoni, 2024-03-18:
Good afternoon, is it possible to add support for the 8BitDo M30 and ather gamepads to Batocera? Doesn't connect :(

Shakyhandsmcg, 2024-03-19:
Does it do a thing where it will connect, start to pair and and then you get a message saying something like 'connection failed. Give up'

stidch, 2025-01-10:
I have the same problem here with multiple controllers. When I follow the procedure in the documentations the pairing always fails with `Connection failed. Give up`

Shakyhandsmcg, 2025-01-10:
> I have the same problem here with multiple controllers. When I follow the procedure in the documentations the pairing always fails with `Connection failed. Give up`

oo this was a blast fromt he past - I managed to get it working eventually - Used a USB cable to connect wired (control still in xinput mode) then once disconnected the cable it seemed to just pair normally via bluetooth - Give it a go if you can!

stidch, 2025-01-11:
I didn't work with neither the 8BitDo nor the PS4 Controller. Now, I just updated to Batocera 41 and suddenly everything is working without any issue. 😅