rg35xx-cfw/rg35xx-cfw.github.io #198: HDMI output issues in 20240306 version

igmarsal, 2024-03-22:
In this latest version the HDMI works but only shows on the screen a third or less of the image. With the previous versions it did not happen

reyaz006, 2024-04-02:
My experience:
- Most of the times the output seems to be a proper 1080p image.
- Other times it's shifted by ~50% up so only ~50% of the bottom is visible.
- Screen rotation setting doesn't seem to do anything. Kind of hoped it could work in vertical mode.
- HDMI Audio output works but all audio can stop for a few seconds by itself, or after spending some time in the RetroArch menu (not sure, it just felt that way).
- After removing HDMI cable the main screen will remain dark until a reboot - of not always then at least most of the times.
- After removing HDMI cable the audio output setting might need manual touch in order for a device to produce audio again.

mtoensing, 2024-06-14:
Is there a chance this gets solved? Because the other distros like Knulli and co have other problems.