rg35xx-cfw/rg35xx-cfw.github.io #200: Colecovision doesn't support Super Game Module (SGM) roms

SmudoTelic, 2024-03-24:
the integrated emulator of colecovision doesn't start roms w/ Super Game Module games. the reason for that ist the BlueMSX don't support the SGM correctly. 
the emulator gearcoleco supports SGM (https://docs.libretro.com/library/gearcoleco/) - but it isn't installed and not selectable in advanced game / emulator-menu. 

would it be possible to implement the gearcoleco emulator in next releases?

acmeplus, 2024-03-28:
Sure, gearcoleco was present at least in a previous version of the firmware. Since there are now major changes expected due to the 64bit updates, I'll check after those updates are completed.

In the meantime here's the core:
[gearcoleco_libretro.so.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/14790091/gearcoleco_libretro.so.zip)

Maxu3x, 2025-08-21:
I recently did an Update_All and  the Colecovision core still does not support games that support Super Game Module.