rg35xx-cfw/rg35xx-cfw.github.io #201: Various Issues and UI Suggestions

rebeltaz, 2024-03-25:
I've tried several UIs for the rg35xx H and this is by far the best, so please don't think that I am complaining. Even at that, though, there are several issues I've come across and a few things I'd like to suggest, if I may.

ISSUES:

- I've seen the issues about the battery draining if you shut down with it plugged in, but I have found that once you've plugged it in to a charger, even if you unplug it before shutting down, it's not going fully into shutdown and is drain the battery by about 20% over six hours or so. I **can** long-press the Power button until the LED goes out to fully it down without battery drain. 
- Additionally, the battery doesn't seem to be charging if I plug the cord in with the unit powered on. The battery icon doesn't change to a charging symbol and the battery level does not increase at all. This is whether the handheld is left with the screen on or with the screen off. It is only charging in Shutdown mode or completely powered off as mentioned above.
- I cannot seem to find any Brightness settings anywhere either in the hotkeys or the menus.
- I cannot seem to find any way to get to RetroArch settings outside of the actual game itself.
- When you select Swap A-B in the settings (which is more natural and user intuitive), this doesn't extend to the RetroArch menus or within any programs like file browser. This is **very** confusing.

SUGGESTIONS:

- Editing or adding Genre and/or Category entries is PAINFUL! To add a Genre entry, you have to go into the Games Option menu; go Up to Edit Metadata; go Up to Genres; scroll through the long, unsorted list of available genres; select one (or more); back out of that menu; back out of that menu; allow it to Save the Metadata. And you have to do that for EACH game. If you are just adding one or two games to your collection, that's not too bad, but if you have 1000+ games (which is not hard) then this is a PAIN. 
.
. It would be MUCH better and easier if you could go to the Category or Genre to which you wanted to add games; select an Add Games menu option that would then allow you to go through the All Games list and long-press the Add to Favorites button to add the selected game to the current Category/Genre. 
.
. I only know about this easier method coming from ArkOS. 

- About Categories... when you press Back in the Main Menu (to go to the Select Manufacturers list), Categories are all grouped as a single entry. For some people (myself included) it's much better to find games based on their Category or Genre (which is why you would bother to add that information in the first place). I have all of the manufactures turned off so that ONLY the Genres are listed. It would be really nice if Collections were not all grouped together in the Manufacturers list but broken out individually when selected to be shown in the settings.

I know that these are small things, but they would make for a much nicer experience.

rooster85, 2024-03-26:
Fyi since batocera overrides the button swap in RetroArch, just add this line to batocera.conf file: global.retroarch.menu_swap_ok_cancel_buttons=true this and a whole lot of other options are described here: https://wiki.batocera.org/advanced_retroarch_settings

rebeltaz, 2024-03-27:
> Fyi since batocera overrides the button swap in RetroArch, just add this line to batocera.conf file: global.retroarch.menu_swap_ok_cancel_buttons=true this and a whole lot of other options are described here: https://wiki.batocera.org/advanced_retroarch_settings

I'll check that out. Appreciate that.

rebeltaz, 2024-04-09:
> Fyi since batocera overrides the button swap in RetroArch, just add this line to batocera.conf file: global.retroarch.menu_swap_ok_cancel_buttons=true this and a whole lot of other options are described here: https://wiki.batocera.org/advanced_retroarch_settings

I've tried this a couple of times, and it doesn't seem to be working for me. I added that line to the "## Advanced RetroArch configuration" section of batocera.conf like so:

```
## Advanced RetroArch configuration
## See https://wiki.batocera.org/advanced_retroarch_settings
#snes.retroarch.menu_driver=rgui
#global.retroarch.input_max_users=4
global.retroarch.menu_swap_ok_cancel_buttons=true
```

That should work, no? I tried setting that flag in the retroarch.cfg file, as well, but of course it was just overwritten.