rg35xx-cfw/rg35xx-cfw.github.io #202: v40 Emulators and Ports crash immediately

6jarjar6, 2024-03-26:
2024-03-26 00:30:08,569 ERROR (unixSettings.py:36):__init__ [Errno 5] Input/output
 error: '/userdata/system/configs/retroarch/retroarchcustom.cfg'
2024-03-26 00:30:08,579 ERROR (emulatorlauncher:606):<module> configgen exception:
Traceback (most recent call last):
  File "/usr/bin/emulatorlauncher", line 604, in <module>
    exitcode = main(args, maxnbplayers)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/emulatorlauncher", line 99, in main
    return start_rom(args, maxnbplayers, args.rom, args.rom)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/emulatorlauncher", line 266, in start_rom
    cmd = generator.generate(system, rom, playersControllers, metadata, guns, whee
ls, gameResolution)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/configgen/generators/libretro/libretroGe
nerator.py", line 70, in generate
    libretroRetroarchCustom.generateRetroarchCustom()
  File "/usr/lib/python3.11/site-packages/configgen/generators/libretro/libretroRe
troarchCustom.py", line 86, in generateRetroarchCustom
    retroarchSettings.write()
  File "/usr/lib/python3.11/site-packages/configgen/settings/unixSettings.py", lin
e 39, in write
    fp = open(self.settingsFile, 'w')
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 5] Input/output error: '/userdata/system/configs/retroarch/retroar
chcustom.cfg'
~
~
~
~
~
~
~
                                                                20,4          All

/bin/sh: 1: batocera-mouse: not found
evmapy: no process found
/bin/sh: 1: batocera-mouse: not found
2024-03-26 00:33:30,659 ERROR (emulatorlauncher:606):<module> configgen exception:
Traceback (most recent call last):
  File "/usr/bin/emulatorlauncher", line 604, in <module>
    exitcode = main(args, maxnbplayers)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/emulatorlauncher", line 99, in main
    return start_rom(args, maxnbplayers, args.rom, args.rom)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/emulatorlauncher", line 266, in start_rom
    cmd = generator.generate(system, rom, playersControllers, metadata, guns, whee
ls, gameResolution)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/configgen/generators/ppsspp/ppssppGenera
tor.py", line 18, in generate
    ppssppConfig.writePPSSPPConfig(system)
  File "/usr/lib/python3.11/site-packages/configgen/generators/ppsspp/ppssppConfig
.py", line 34, in writePPSSPPConfig
    with open(ppssppConfig, 'w') as configfile:
         ^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 5] Input/output error: '/userdata/system/configs/ppsspp/PSP/SYSTEM
/ppsspp.ini'
~
~
~
~
~
~
~
~
~
~
~
~
                                                                20,4          All

Apologies for formatting I am using a phone

acmeplus, 2024-03-28:
This looks like corruption, the input output errors look like the system is trying to access some files but cannot get them. If you are using 1 SDCARD, try to rename/move the system folder from the last partition and try again.

6jarjar6, 2024-03-29:
@acmeplus I am using 2 SD cards. And I reflashed v40 and got the same error, using Balena. Is there software you'd recommend for checking to see if there's a bad sector on the SD card? I have a good SD card Samsung evo.

WesRichardson33, 2024-04-09:
Thanks, I'm guessing I did that by coping things over to another card. I will keep this in mind if it happens again