rg35xx-cfw/rg35xx-cfw.github.io #205: Is there a tutorial to build kernel for rg35xxh?

KatyushaScarlet, 2024-03-28:
Hi,

I discovered that there is a prebuilt kernel located in board/batocera/allwinner/h700/rg35xx-plus. It appears that this cfw created for the rg35xxh uses this uImage instead of building it from buildroot.

Currently, I am attempting to build the latest mainline kernel for the rg35xxh. I have followed the offical batocera's toturial, and also tried some tutorials on building the mainline kernel for Orange Pi Zero2/Zero3, but both of these methods not working for my rg35xxh. I'm wondering if there is a tutorial or guidance available for building the kernel for rg35xxh?

Thanks!

acmeplus, 2024-03-28:
Yes, the kernel and bootloader included are the ones from the stock. Unfortunately, anbernic has not published the modified kernel and bootloader sources. 

However, if you want to compile the kernel and bootloaders that were (most likely) used as the base for the stock firmware, you can have a look to the orange build https://github.com/orangepi-xunlong/orangepi-build, and you can use their build.sh script to compile both the kernel and uboot. For the kernel use the branch orange-pi-4.9-sun50iw9

For mainline, there's currently an ongoing effort by the linux sunxi team and many others to try to get mainline to work for the H700. Currently we have mainline uboot booting up (https://git.sr.ht/~tokyovigilante/u-boot) and you can also get the latest mainline kernel to work with some patches (use lore to get them):
* 20240310_andre_przywara_regulator_add_x_powers_axp717_pmic_support
* v2_20240318_andre_przywara_cpufreq_sun50i_add_allwinner_h616_support
* v5_20240219_andre_przywara_add_support_for_h616_thermal_system

With that you can get a system booting with bluetooth, wireless, but no GPU or internal display. I'm sure things will get fixed in the few weeks/months and we will have a full open source solution for these boards.

KatyushaScarlet, 2024-03-28:
> Yes, the kernel and bootloader included are the ones from the stock. Unfortunately, anbernic has not published the modified kernel and bootloader sources.
>
> However, if you want to compile the kernel [...]

Thank! This is really a huge help.

loki666, 2024-03-29:
so, using https://git.sr.ht/~tokyovigilante/u-boot and a mainline kernel (ie: 6.8.2) + patches using buildroot should get you a uboot + kernel + ramdisk + rootfs ?

how about building the final img? like all these blob partitions (boot0 and whatnot) included in batocera ?
can we just reuse them?

what's the genimage config ?

you said GPU and internal LCD are dead, what about HDMI ? or is it missing the whole sunxi disp driver ?

thanks

loki666, 2024-03-30:
do you have the kernel lore urls for the patches,

I can only find https://lore.kernel.org/lkml/20240318115023.59e368f2@donnerap.manchester.arm.com/

thanks

acmeplus, 2024-03-30:
No, the orange pi zero3 DT won't work, use the one from the u-boot repo I posted above.

For the patches, search in google those patches + lore, and then use b4 to download the patch sets.

The mainline kernel does not need anything from the stock, so you don't need boot0, boot-package, etc. It's all self contained. And yes, the HDMI is working with the jernejsk H616/618 hdmi patches.

About an image, you can use the same recipe and genimage as the batocera orange pi zero 2 / H616.

loki666, 2024-03-30:
ok so the uboot dtb is complete enough for the kernel ?
I'll check how to config Buildroot, to use uboot dtb for kernel

I used b4 for the patch I found, but couldn't found the others... 

I'm currently trying with bare buildroot

thanks for the details

loki666, 2024-03-31:
ok so I had a "hooray" moment, when the led turned green with my buildroot image
ofc nothing else, no hdmi/lcd but that was kind of expected... trying to get the wifi up so I can ssh into it

but now, it doesn't want to boot batocera or stock sd card....

any idea? or am I good to wire my ft232 to the uart port of the board?

KatyushaScarlet, 2024-03-31:
> Yes, the kernel and bootloader included are the ones from the stock. Unfortunately, anbernic has not published the modified kernel and bootloader sources.
>
> However, if you want to compile the kernel [...]

Hi,
I have tried to build the kernel and u-boot as you mentioned, but now I'm stuck at the booti command in u-boot.
I have patched the kernel 6.9-rc1, and use defconfig to build the kernel.
Do you have any idea where I might have gone wrong?
Thanks!

Here's my output from rg35xxh uart:
```
U-Boot SPL 2024.04-rc3-g8e955e273b-dirty (Mar 31 2024 - 11:27:02 +0800)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0(debug):v2.10.0-663-g79da34891
NOTICE:  BL31: Built : 10:30:03, Mar 31 2024
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a090a70, model: Anbernic RG35XX Plus
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for erratum 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for erratum 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2024.04-rc3-g8e955e273b-dirty (Mar 31 2024 - 11:27:02 +0800) Allwinner Technol

CPU:   Allwinner H616 (SUN50I)
Model: Anbernic RG35XX Plus
DRAM:  1 GiB
Core:  74 devices, 18 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0, mmc@4021000: 1, mmc@4022000: 2
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Card did not respond to voltage select! : -110
No EFI system partition
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
Loading Boot0000 'mmc 0' failed
EFI boot manager: Cannot load any image
No ethernet found.
missing environment variable: pxeuuid
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
Retrieving file: pxelinux.cfg/000000
No ethernet found.
Retrieving file: pxelinux.cfg/00000
No ethernet found.
Retrieving file: pxelinux.cfg/0000
No ethernet found.
Retrieving file: pxelinux.cfg/000
No ethernet found.
Retrieving file: pxelinux.cfg/00
No ethernet found.
Retrieving file: pxelinux.cfg/0
No ethernet found.
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
No ethernet found.
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
No ethernet found.
=> setenv bootargs 'console=ttyS0,115200 earlycon'
=> fatload mmc 0:1 0x4fa00000 sun50i-h700-anbernic-rg35xxph.dtb
137957 bytes read in 7 ms (18.8 MiB/s)
=> fatload mmc 0:1 0x40200000 Image
44296704 bytes read in 1839 ms (23 MiB/s)
=> booti 0x40200000 - 0x4fa00000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
Working FDT set to 4fa00000
   Loading Device Tree to 0000000049fdb000, end 0000000049fffae4 ... OK
Working FDT set to 49fdb000

Starting kernel ...
```

loki666, 2024-03-31:
I haven't wired my uart yet, so I may be stuck at the same step as you...

Does your led turn green?

loki666, 2024-03-31:
@KatyushaScarlet does your RG35XX still boot stock SD card?

loki666, 2024-03-31:
Ok unplugging the battery and back did the trick, it's booting the stock SD card again.

Seems like something is staying in some memory, any idea what? Could it be this TFA thing? Or does the CPU stay in Aarch64 mode?

KatyushaScarlet, 2024-03-31:
> Ok unplugging the battery and back did the trick, it's booting the stock SD card again.
> 
> Seems like something is staying in some memory, any idea what? Could it be this TFA thing? Or does the CPU stay in Aarch64 mode?

1. No, the LED didn't turn green for me.
2. Yes, my RG35XX still boot through stock SD card.
3. Maybe you can try the reset button to force it reboot from another ROM.

loki666, 2024-03-31:
I'll publish my buildroot repo later

loki666, 2024-03-31:
@KatyushaScarlet  here is my branch https://github.com/loki666/buildroot/tree/rg35xx-plus , you can give it a shot

KatyushaScarlet, 2024-04-01:
> @KatyushaScarlet here is my branch https://github.com/loki666/buildroot/tree/rg35xx-plus , you can give it a shot
It works!
Here's my output of the image built from your buildroot:
```
U-Boot SPL 2024.04-rc3 (Apr 01 2024 - 20:11:03 +0800)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0	(release):v2.10
NOTICE:  BL31: Built : 20:09:22, Apr  1 2024
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0905f8, model: Anbernic RG35XX Plus
ERROR:   RSB: set run-time address: 0x10003

U-Boot 2024.04-rc3 (Apr 01 2024 - 20:11:03 +0800) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: Anbernic RG35XX Plus
DRAM:  1 GiB
Core:  79 devices, 18 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0, mmc@4021000: 1, mmc@4022000: 2
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
1:	default
Retrieving file: /boot/Image
append: root=PARTUUID=c16fc210-4bd5-4ae9-9887-1b80b9cdd5bd rootwait console=ttyS0,115200 rootfstype=ext4 quiet panic=10
Retrieving file: /boot/allwinner/sun50i-h700-anbernic-rg35xxplus.dtb
** File not found /boot/allwinner/sun50i-h700-anbernic-rg35xxplus.dtb **
Skipping fdtdir /boot for failure retrieving dts
Moving Image from 0x40080000 to 0x40200000, end=42c50000
## Flattened Device Tree blob at 7bf32050
   Booting using the fdt blob at 0x7bf32050
Working FDT set to 7bf32050
   Loading Device Tree to 0000000049ff5000, end 0000000049fffd51 ... OK
Working FDT set to 49ff5000

Starting kernel ...

[    0.586234] debugfs: Directory 'vcc-1v8' with parent 'regulator' already present!
[    1.824527] rtw_8821cs mmc1:0001:1: failed to request firmware
[    1.837118] rtw_8821cs mmc1:0001:1: failed to load firmware
[    1.842829] rtw_8821cs mmc1:0001:1: failed to setup chip efuse info
[    1.849131] rtw_8821cs mmc1:0001:1: failed to setup chip information

Welcome to Anbernic RG35XX Plus/H
buildroot login: root
# uname -a
Linux buildroot 6.8.2 #1 SMP PREEMPT Mon Apr  1 20:18:35 CST 2024 aarch64 GNU/Linux
```

loki666, 2024-04-01:
@KatyushaScarlet 
Nice if you could try to make the wifi an ssh working it would be great

Because without the uart I can't do much

KatyushaScarlet, 2024-04-01:
> @KatyushaScarlet Nice if you could try to make the wifi an ssh working it would be great
> 
> Because without the uart I can't do much
> 
> It looks like it can't find the firmware... Maybe I missed something

Yes, the wifi seems work, but I can not connect to any AP for now.
Also, you have to do some tweaking:

1. Put the device tree to `/boot/allwinner/sun50i-h700-anbernic-rg35xxplus.dtb`.
2. Put the firmware to `/lib/firmware/rtw88/rtw8821c_fw.bin`, which can be download from 'https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw88/rtw8821c_fw.bin'
3. Run `ifconfig wlan0 up`

loki666, 2024-04-01:
ok let me try to patch my repo for this

KatyushaScarlet, 2024-04-01:
> ok let me try to patch my repo for this
> 
> @KatyushaScarlet the current image include iwd, you just need to drop a file in `/var/lib/iwd/<SSID>.psk`
> 
> with a content like
> 
> ```
> [Security]
> Passphrase=CHANGE_ME
> 
> [IPv4]
> Address=192.168.1.21
> Netmask=255.255.255.0
> Gateway=192.168.1.1
> Broadcast=192.168.1.255
> DNS=192.168.1.1
> ```

Thanks!
The wifi is work for me, and I can get access to the shell through ssh.

```
# iwconfig
lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"<SSID>"  
          Mode:Managed  Frequency:5.26 GHz  Access Point: <MAC>   
          Bit Rate=433.3 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=70/70  Signal level=-32 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:1   Missed beacon:0
```

loki666, 2024-04-01:
can you boot back to stock/batocera SD card ?

KatyushaScarlet, 2024-04-01:
> can you boot back to stock/batocera SD card ?

Yes I can, no issue.

KatyushaScarlet, 2024-04-01:
> ok nice,
> 
> you can try the latest version of my buildroot, it should take care of the wifi firmware and move the the dtb to the correct folder

That seems to solve the problem, thanks!

loki666, 2024-04-01:
ok, let's try to get the LCD with a simple frambuffer working
I found a ref to a LCD replacement part `WL-355608-A8` which also seems to be used for the RG350M

@acmeplus do you know what's missing for the LCD, is it at the dtb level, or the kernel level, or both ?

loki666, 2024-04-01:
@KatyushaScarlet I'm still not able to boot back stock/batocera SD card?
how did you shutdown the device while in the buildroot image?
I tried with a long press on the power button, but then stock sd card doesn't boot, but I can stil boot buildroot image

KatyushaScarlet, 2024-04-01:
> @KatyushaScarlet I'm still not able to boot back stock/batocera SD card? how did you shutdown the device while in the buildroot image? I tried with a long press on the power button, but then stock sd card doesn't boot, but I can stil boot buildroot image

I just unpluged the battery and power cable to shutdown the device.
Then, I replace the MicroSD card with stock/batocera, plug in the power cable, it boot as usual.

loki666, 2024-04-01:
arf, yes that also works for me, but not ideal
@acmeplus any idea what's blocking stock OS to boot ?

loki666, 2024-04-01:
@KatyushaScarlet could you post the full UART log of the boot process ?

acmeplus, 2024-04-01:
> @KatyushaScarlet could you post the full UART log of the boot process ?

Look at my gists, it's should be all there

acmeplus, 2024-04-01:
> arf, yes that also works for me, but not ideal @acmeplus any idea what's blocking stock OS to boot ?

I've not done anything with this in several days, but note that I'm always testing these things with the UART connected and without battery. That simplifies things a lot. About the system not booting again with stock, the only thing I can think off is that is not really off. Try to use a call to /proc/sysrq to shut it down completely.

loki666, 2024-04-01:
ok I'll give a shot at /prioc/sysrq
any hint on what's going on with the LCD ?

loki666, 2024-04-01:
`echo o > /proc/sysrq-trigger` didn't work, still no stock os boot

loki666, 2024-04-02:
@acmeplus how to enable HDMI?

loki666, 2024-04-02:
> It works!
> Here's my output of the image built from your buildroot:
> ```
> U-Boot SPL 2024.04-rc3 (Apr 01 2024 - 20:11:03 +0800)
> DRAM: 1024 MiB
> Trying to boot from MMC1
> NOTICE:  BL31: v2.10.0 [...]

Was the kernel able to boot without moving the dtb in /boot/allwinner ?

KatyushaScarlet, 2024-04-02:
> Was the kernel able to boot without moving the dtb in /boot/allwinner ?

yes.

KatyushaScarlet, 2024-04-23:
Hi, I found a repo for kernel 5.15 that includes Allwinner's BSP with a number of Allwinner out-of-tree drivers. The BSP supports many Allwinner SoCs, such as sun50iw9 (H700, for RG35XXH) and sun50iw10 (A133, for Trimui Smart Pro). I think you might be interested.
@acmeplus @loki666 
https://github.com/chainsx/linux-t527

loki666, 2024-04-24:
nice, but I don't think it will brings much more on table than the precompiled 4.9 we already have
and without a dts for RG35XX, it wont be of much help

you can try to repack it and boot using the recipes here
https://knulli.org/guides/h700-firmware-extract/

KatyushaScarlet, 2024-04-25:
> nice, but I don't think it will bring much more on table than the precompiled 4.9 we already have and without a dts for RG35XX, it wont be of much help
> 
> you can try to repack it and boot using the recipes here https://knulli.org/guides/h700-firmware-extract/

Oh ,does the device tree needed by kernel 5.15 different from kernel 4.9 or kernel 6.9 ? I don't know much about it. I think that at least we may have an opensource solution for the H700 and A133 devices.