rg35xx-cfw/rg35xx-cfw.github.io #207: Batocera v40 does not create a SHARE partition

NeoFax75, 2024-03-30:
There is no share partition created viewable in windows 11.

uzumak11, 2024-03-30:
Pop it into the device and boot after flashing. Share drive is created on device

NeoFax75, 2024-03-30:
I did that, but on windows the share drive does not show up.  However it does on linux.

uzumak11, 2024-03-30:
You will have to assign a drive letter on windows for it to appear as removable media. Go to disk management > assign drive letter

NeoFax75, 2024-03-30:
Thank you that worked great. On previous builds it just came up.

Lokesh-Karthik, 2024-10-31:
Hey there! I seem to run into the same issue where, after flashing to a USB drive the share is created but after plugging to a different PC for batocera it doesn't create any share partition

Lokesh-Karthik, 2024-10-31:
ok i had to clone the share partition from the USB drive to the HDD which had unallocated space with batocera 
and the next time I booted it started resizing (from 256mb to 345GB) and booted well