rg35xx-cfw/rg35xx-cfw.github.io #210: Sega Pico

Sp33dFr34k, 2024-04-02:
There is a Sega Pico folder and I can scrape for it, but it doesn't show up at all, there's no emulator for it?

Using Batocera V40 and RG35XX H.

probe0580, 2024-04-03:
Rom Folder = pico8
File Extension = .p8 (not zip)

Sp33dFr34k, 2024-04-03:
> Rom Folder = pico8 File Extension = .p8 (not zip)

Sega Pico != Pico 8 :)

From the pico folder's _info.txt:

> ## SYSTEM SEGA PICO ##
-------------------------------------------------------------------------------
ROM files extensions accepted: ".bin .md .zip .7z"

The folder I use is pico, and the extensions are .zip. Within the .zip there are .md files. The games run fine via the All Games collection as mentioned above, so I think this must be something simple.

uzumak11, 2024-04-03:
Sega pico is not mentioned as one of the supported systems under release notes ("Cores/Emulators status")

The folders are populated/created cos that's how batocera(main) builds it

Sp33dFr34k, 2024-04-03:
> Sega pico is not mentioned as one of the supported systems under release notes ("Cores/Emulators status")
> 
> The folders are populated/created cos that's how batocera(main) builds it

Ah okay, didn't check that. Thanks. The emulator seems to be there though as the games work. But I'll close it then as it's not a bug.

closed 2024-04-03 by Sp33dFr34k