rg35xx-cfw/rg35xx-cfw.github.io #212: Network connection issues

ABFL94, 2024-04-02:
Somehow my RG35xx plus is not able to find any networks I tried with my WIFI it won't work, I even wiped that sd card clean and did the whole process again. I even tried using my mobile hotspot and it won't search for any networks.

One more issue to add,

 When turning off the system and booting it back on, I have to set up all of my buttons again it somehow doesn't remember the button layout anymore.

Sp33dFr34k, 2024-04-03:
I also have wifi connectivity issues. Using Unifi U6 access points, WPA2 and PMF is disabled. Disabling/enabling wifi works sometimes, but it's quite finicky. Not sure what's going on there. Also, I'd log your second issue separately.

uzumak11, 2024-04-03:
WPA3 is the one that doesn't work. 

WPA2 encryption for WiFi works fine

Sp33dFr34k, 2024-04-03:
> WPA3 is the one that doesn't work.
> 
> WPA2 encryption for WiFi works fine

That's why I mentioned I'm on WPA2 with no PMF and it doesn't work flawlessly. Only sometimes, when disabling/enabling it works.

uzumak11, 2024-04-03:
Aaah I see, I read that as wpa2 disabled for some reason. 

The WiFi range on the device is weak, so I wouldn't discount that as contributing to your issue.

Did you try both 5ghz and 2.4ghz?

uzumak11, 2024-04-03:
>  When turning off the system and booting it back on, I have to set up all of my buttons again it somehow doesn't remember the button layout anymore. 

This seems to happen a lot if your share drive is fat32. Exfat is better. 

Even with that, you'll have to look at Linux file permissions being intact.

Easy fix is to delete/move your system folder, so the OS can recreate this

MrFobs1, 2024-06-09:
I created an account to comment this. I found with my Ubiquiti router It had to be WPA2 AND it could not be a hidden network. I switched my guest network to broadcast is SSID and now WiFi works no problem. 

You also may need to remove the password, press back, then enter the password again for the change to be accepted on the RG35XXP