rg35xx-cfw/rg35xx-cfw.github.io #214: Libretro: mupen64plus-next Emulator unable to access RetroArch Menu

milktoastrat, 2024-04-03:
So far the N64 is the only system I have found with this issue. I have tried several fixes to no avail. I have discussed on reddit and it is a common problem with this release. Even the notifications that pop up for retroacheivements are the same unreadable gradient. I have included a photo of what happens when holding hotkey+B

[image]

uzumak11, 2024-04-03:
Add the following line to batocera.conf

global.retroarch.video_driver = "gl"

SaltyRum, 2024-04-15:
I'm having a similar issue and the gl line hasn't fixed it. I'm not getting a full rainbow screen just borders and when I try to launch retroach the screen changes colour but I can see the options. Also my N64 games are running at about 5fps