rg35xx-cfw/rg35xx-cfw.github.io #216: Problem displaying the correct battery status

cfboehm, 2024-04-04:
At some point the battery status is freezing in EmulationStation.

Example:
- battery status is at 90%
- the battery drains 15%
- status in ES is still 90%
- in RetroArch the status is correct at 75%

I have not yet been able to find out when this will happen. I suspect the hibernation mode. When it's happening, the WiFi indicator is also frozen. A restart fixing the problem. I'am using version 20240306 (v40) on a Anbernic RG35xx+.

LuisLopes94, 2024-04-04:
Seeing the same issue as well. On a RG35XX+ as well.

Issue happens even if I don't hibernate the console. Just by playing a game and then exiting it, I only get the correct battery if I reboot the device.

seanstardev, 2024-04-05:
Confirmed to happen on my Plus as well, with latest Bato version.

ToyKeeper, 2024-04-07:
Same issue here.  ES-DE's battery info just doesn't update very often.  It's sometimes hours out of date, while every other program shows the correct status.  One of the only ways I've found to make it update the battery info is to let the console sit idle for like 30+ minutes, then press a button to wake it up... and even that isn't very reliable.  Otherwise, I have to go into retroarch to see battery status, or ssh in and run btop, or otherwise find some other program with working battery info.

GiamBoscaro, 2024-06-08:
> Seeing the same issue as well. On a RG35XX+ as well.
> 
> Issue happens even if I don't hibernate the console. Just by playing a game and then exiting it, I only get the correct battery if I reboot the device.

I have the same problem. I never know the real battery status until I reboot.