rg35xx-cfw/rg35xx-cfw.github.io #220: Unable to play gab room hack

Rcleto, 2024-04-05:
Playing the latest release on and RG35XXH and i am unable to play gab room hacks like pokemon unbound or last fire red. The game just does not boot. 
I tryied several rooms hacks and any of them works. Already tried selecting the GBA emulator and neither MGBA or VBA-M works.  Also download all the links. 
The rooms work well on Delta on my iPhone so the rooms and patched correctly

uzumak11, 2024-04-05:
Playing unbound flawlessly on v40

Rcleto, 2024-04-05:
Really??
Did you switch any settings?
Which emulator are you using?

Rcleto, 2024-04-05:
> Playing unbound flawlessly on v40

But are you using a patched rom or patching in the system?

uzumak11, 2024-04-05:
MGBA.

I think your rom source is corrupt.

Where are you getting your hacks from?

The best way is to patch it yourself, go to unbound discord, download the nups file, and patch it to the source file they ask for (a copy of emerald, a specific one)

Rcleto, 2024-04-06:
But I tried with dozens of gba room hack not just unbound.
And these room works perfectly on Delta on my iPhone. 
Must be something missing in the settings... Did you change anything?

uzumak11, 2024-04-06:
Nothing. Stock settings. I'm using mgba emulator if the helps

Rcleto, 2024-04-06:
Well sorted it out...
The rooms play well on gba but not on zip...
Probably because zip on mac... 

Well thank you for all the trouble