rg35xx-cfw/rg35xx-cfw.github.io #221: Games.com noit appearing

MattyH81, 2024-04-06:
Tried adding roms both as .zip and .bin.  system does not show in menu.

uzumak11, 2024-04-06:
Which system are you referring to?

MattyH81, 2024-04-06:
It's in the title.  Games.com (tiger)

uzumak11, 2024-04-06:
https://wiki.batocera.org/systems:gamecom

Not sure about support on our device though. 

It's not part of the release notes under "Cores/Emulators status"

MattyH81, 2024-04-06:
Yeah. I saw the folder had been added do thought I'd try.  I'll leave roms for there now in case it gets added.

uzumak11, 2024-04-06:
The folders are part of batocera main (linux). 

The release notes are a better indicator of what is currently supported, so be sure to check that.

Having said that, the dev is planning to merge our device build with batocera main rather soon, which would hopefully add support for all the extra systems.

I should check out this system though, never heard of it growing up!