rg35xx-cfw/rg35xx-cfw.github.io #222: No gpsp libretro core?

Qaeoss, 2024-04-07:
I only see mgba and vba available as cores for gba, is gpsp available to use as well or is there a way I can add it?

uzumak11, 2024-04-07:
gpsp was removed by batocera main, as it's considered useful only for extreme low end hardware. 

Mgba works great on here.

Dev mentioned he'll consider adding it back in for a future update

Qaeoss, 2024-04-08:
Oh, no complaints about mgba I just prefer gpsp when grinding as it fast forwards much… faster.

uzumak11, 2024-04-08:
I remember reading something about increasing fast forward performance on mgba.

I'll link it here when I find it.

Goldensun7, 2024-04-15:
I am using STOCK OS last update .... and I run my pokemonn games on  GPSP.. should I use MGBA??? whats the difference and also how do I fast forward on mgba? in pokemon games
I am not running batocera, just stock os

import-tiago, 2024-04-24:
mgba are current missing the color correction feature in Batocera v40. gpsp could be a good alternative in this case...