rg35xx-cfw/rg35xx-cfw.github.io #223: [Feature request] One tap shutdown and autosave and autoload on boot

harun911, 2024-04-07:
One of the things I love about my RG35XX with Garlic OS or Miyoo Mini's, or my RG353V with ArkOS, is the ability to instantly shutdown the device with one tap. And immediately continue playing when I boot it back up. Would it be possible to implement something like this for the RG35XX plus? Even if it's only for Retroarch cores (that's the limitation on the RG353V aswell). It would make it that much more of a pickup and play whenever wherever kind of a device.
Thanks for all your hard work so far!

harun911, 2024-04-09:
I agree its not so bad, but it would be a nice quality of life enhancement! 😄

kraptor, 2024-04-10:
@ToyKeeper @harun911 It's even faster :)
1. Hit hotkey to manually save state.
2. Hit simultaneously **Fn + On/Off** button (immediate turns off the device).
3. Later, restart game you were playing and load gamestate.

synak, 2024-04-10:
RetroArch itself has options to for "Auto Save State" (upon closing content) and "Load State Automatically", which gets you most of the way there, if I'm understanding what you're trying to replicate.

Settings -> Saving -> Auto Save State
Settings -> Saving -> Load State Automatically

harun911, 2024-04-10:
> @ToyKeeper @harun911 RetroArch itself has options to for "Auto Save State" (upon closing content) and "Load State Automatically", which gets you most of the way there, if I'm understanding what you're trying to replicate.
> 
> Settings -> Saving -> Auto Save State Settings -> Saving -> Load State Automatically

Almost, but not quite, yes the feature I'm mentioning relies on the auto save and auto load feature of retroarch, with the difference of being tied to a single button shutdown. I just learned from kraptor, that function button + shutdown button is a thing, didn't knew that before, so that already makes it somewhat better. But the key difference is, automatically saving state when doing a single button shutdown. And when booting back up, booting directly back in the game you've shutdown in, and automatically loading the state. I've tried altering the retroarch settings to enable autosave and autoload btw, but somehow the retroarch config doesn't persist when I save the configuration it. Any idea if I'm missing something?

synak, 2024-04-10:
> I've tried altering the retroarch settings to enable autosave and autoload btw, but somehow the retroarch config doesn't persist when I save the configuration. Any idea if I'm missing something?
Try checking Settings -> Configuration -> Save Configuration on Quit and ensure it's enabled.

synak, 2024-04-14:
> I've tried altering the retroarch settings to enable autosave and autoload btw, but somehow the retroarch config doesn't persist when I save the configuration.

I was previously answering without knowing that batocera overlays retroarch configurations on boot with its own defaults. You may want to check the wiki for advanced settings, including editing batocera.conf which establishes some global setting overrides. [wiki link](https://wiki.batocera.org/batocera_conf_syntax)

For example, I wanted to tweak the rewind configuration and was able to do so with the following commands in batocerf.conf
`global.retroarch.rewind_buffer_size=100
global.retroarch.rewind_buffer_size_step=5
global.retroarch.rewind_granularity=3
`