rg35xx-cfw/rg35xx-cfw.github.io #237: Discharging when powered off and not actively charging

Zach1812, 2024-04-16:
I have noticed that when I plug-in my RG35XX-H, that is turned off, it will charge up and show 100%. Then it'll show that the system is Discharging. If I unplug the system, ensuring the system is off, and leave it for a few hours, the battery level has dropped and the status still says Discharging. Currently, If the system is off and not plugged in, it will lose all charge after 2 days.

Is there a process running in the background even when the system is off? And if so, can it be turned off, or at least draw as little current as possible?

uzumak11, 2024-04-16:
Switching to ext4 reduces drain to 0%

Zach1812, 2024-04-16:
So, if I changed the file system type to ext4 on the Batocera side it would fix the drain? 

1. How would I go about making this happen? Is it a command line thing or would a gui be able to walk me through it. 
2. How would I do this without messing up the SHARE side?

Thanks

uzumak11, 2024-04-16:
It's a complete format of your card, so transferring stuff to PC, formatting, transfer back.

Ext4 is not accessible of windows/mac as well by default. There are third party tools that you can use to access it (paragon). Network share can also be used to transfer files back. I have linux on my PC, so makes things easier.

Zach1812, 2024-04-17:
Ok, I managed to change the boot to ext4 but have still noticed the drain. One other test did produce an interesting result. If it is charging while on, but I remove the power cord before shutting it down, the drain stops. If I plug power back in, the charge continues, but then once removed it starts draining again. It seems to be tied to the amount of power the power charge screen draws. (That shouldn't be a great amount, but seemingly is)

uzumak11, 2024-04-17:
It's not boot, it's the share drive that you have to change to ext4. 

For a single card, that'd be the /userdata partition. 

For a 2 card setup, TF2 has to be made ext4.

I did a write up on Reddit if you'd like to read more about this: https://www.reddit.com/r/RG35XX_Plus/s/AKzKYcup83

Zach1812, 2024-04-17:
Ahh! That's the part I was missing. Sadly I didn't see your post on Reddit. Thanks for posting about it. I thought it was only the boot portion that needed to be Ext4, not the entire card. You are correct that accessing EXT4 is difficult with non-linux based machines. Since acme already knows of the problem, I'll close this thread out.

Thanks!

closed 2024-04-17 by Zach1812