rg35xx-cfw/rg35xx-cfw.github.io #247: anbernic rg35xx plus dapd and wifi

ibriz25, 2024-07-01:
i have anbernic rg35xx plus  i can play games like  gba  but when i try to play psp games like GOD OF WAR that time dpad is not working  
but all buttons are working , and another issue is that when i try to connect wifi to the device (rg35xx plus ) it connect ,even i try to  connect open  wifi connection also  but still its notworking , anyone can tell me how to connect using wpa supplicant file modification

ffleandro, 2024-09-30:
Are you using v40 for the wifi issue?
I think there was a Wifi/Bluetooth issue on v39.

I'm using v40 on the RG35xx Plus and everything is working fine.
Can't comment on the PSP issue since I don't use it.

ffleandro, 2024-09-30:
Link to the [v40 Release](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/releases/tag/rg35xx_plus_h_batocera_20240306)